ManuLST/Marlin4ST #2: how Building the firmware with the OpenSTM32 IDE (IAR)

huixiesui, 2016-10-08:
How to import the IAR version?

"The firmware comes with the complete source code for the OpenSTM32 (free) and IAR
development environments."

I use OpenSTM32  import the Marlin4ST project errors.

St3dPrinter, 2016-10-10:
@huixiesui , can you send the console log?
Did you correctly set the workspace path ? It should be stm32_cube, because all path in the project are relative path to stm32_cube.

huixiesui, 2016-10-10:
Thank you for your reply。

After import errors, I'm going to try to change the operating system test.（I think may be the result of AC6 IDE error.）

The following report will be sent to Eclipse:

------
STATUS
------
pluginId            org.eclipse.cdt.core
pluginVersion       5.11.0.201602051005
code                0
severity            4
message             Error: required build system is not installed
fingerprint         821e369e579e156f14adb0979793c9a6

Exception:org.eclipse.core.runtime.CoreException: required build system is not installed
	 at org.eclipse.cdt.internal.core.settings.model.ExceptionFactory.createCoreException(ExceptionFactory.java:22)
	 at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProvider(CProjectDescriptionManager.java:1190)
	 at org.eclipse.cdt.internal.core.settings.model.CConfigurationDescriptionCache.loadData(CConfigurationDescriptionCache.java:135)
	 at org.eclipse.cdt.internal.core.settings.model.CProjectDescription.loadDatas(CProjectDescription.java:196)
	 at org.eclipse.cdt.internal.core.settings.model.xml.XmlProjectDescriptionStorage.loadProjectDescription(XmlProjectDescriptionStorage.java:500)
	 at org.eclipse.cdt.internal.core.settings.model.xml.XmlProjectDescriptionStorage.getProjectDescription(XmlProjectDescriptionStorage.java:237)
	 at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescriptionInternal(CProjectDescriptionManager.java:437)
	 at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:419)
	 at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:413)
	 at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:406)
	 at org.eclipse.cdt.internal.core.model.CProject.computeSourceRoots(CProject.java:584)
	 at org.eclipse.cdt.internal.core.model.CProject.computeChildren(CProject.java:605)
	 at org.eclipse.cdt.internal.core.model.CProject.buildStructure(CProject.java:569)
	 at org.eclipse.cdt.internal.core.model.Openable.generateInfos(Openable.java:261)
	 at org.eclipse.cdt.internal.core.model.CElement.openWhenClosed(CElement.java:427)
	 at org.eclipse.cdt.internal.core.model.CElement.getElementInfo(CElement.java:305)
	 at org.eclipse.cdt.internal.core.model.CElement.getElementInfo(CElement.java:295)
	 at org.eclipse.cdt.internal.core.model.Parent.getChildren(Parent.java:55)
	 at org.eclipse.cdt.internal.core.model.CProject.getSourceRoots(CProject.java:467)
	 at org.eclipse.cdt.internal.core.model.CModelManager.create(CModelManager.java:337)
	 at org.eclipse.cdt.internal.core.model.CModelManager.create(CModelManager.java:265)
	 at org.eclipse.cdt.internal.core.model.DeltaProcessor.createElement(DeltaProcessor.java:87)
	 at org.eclipse.cdt.internal.core.model.DeltaProcessor.traverseDelta(DeltaProcessor.java:453)
	 at org.eclipse.cdt.internal.core.model.DeltaProcessor.traverseDelta(DeltaProcessor.java:471)
	 at org.eclipse.cdt.internal.core.model.DeltaProcessor.processResourceDelta(DeltaProcessor.java:433)
	 at org.eclipse.cdt.internal.core.model.CModelManager.resourceChanged(CModelManager.java:900)
	 at org.eclipse.core.internal.events.NotificationManager$1.run(NotificationManager.java:299)
	 at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	 at org.eclipse.core.internal.events.NotificationManager.notify(NotificationManager.java:289)
	 at org.eclipse.core.internal.events.NotificationManager.broadcastChanges(NotificationManager.java:152)
	 at org.eclipse.core.internal.resources.Workspace.broadcastPostChange(Workspace.java:373)
	 at org.eclipse.core.internal.resources.Workspace.endOperation(Workspace.java:1470)
	 at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:46)
	 at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

------
STATUS
------
pluginId            org.eclipse.cdt.core
pluginVersion       5.11.0.201602051005
code                0
severity            4
message             required build system is not installed [detached from CoreException of Status 'Error: required build system is not installed' by Error Reporting]
fingerprint         04927bcdd8468e15312afc06d92ade68

------
REPORT
------
anonymousId         cf97e692-be11-40b0-ba43-640129c8f684
name                
email               
comment             
eclipseBuildId      4.5.2.M20160212-1500
[... 316 lines cut ...]
version             1.100.0.v20160217-0435

name                org.eclipse.equinox.app
version             1.3.300.v20150423-1356

name                org.eclipse.equinox.launcher
version             1.3.100.v20150511-1540

name                org.eclipse.jface
version             3.11.1.v20160128-1644

name                org.eclipse.swt
version             3.104.2.v20160212-1350

name                org.eclipse.ui
version             3.107.0.v20150507-1945

name                org.eclipse.ui.forms
version             3.6.200.v20150506-2029

name                org.eclipse.ui.ide.application
version             1.1.0.v20150422-0725

name                org.eclipse.ui.ide
version             3.11.0.v20150825-2158

St3dPrinter, 2016-10-10:
Just a question: did you use the AC6 IDE installer version or did you added AC6 plugin in an existing version of Eclipse?

huixiesui, 2016-10-10:
Downloaded from the http://www.openstm32.org/ version “install_sw4stm32_win_64bits-v1.8.exe ”

“install_sw4stm32_win_64bits-latest”

St3dPrinter, 2016-10-10:
Ok, thank you.
I will test it again.

huixiesui, 2016-10-10:
After the existing Eclipse version add plug-ins can be compiled.
Thanks a lot.

closed 2016-10-10 by huixiesui

St3dPrinter, 2016-10-11:
Hi,

Yesterday evening, I've tested again the complete setup on Windows 10.
I used this installer:  install_sw4stm32_win_64bits_v1.8

Once installed:
- Set the workspace to stm32_cube
- Import  ->  General  ->  Existing Projects into workspace

Right click on the project name -> Refresh
Right click on the project name -> Build Project

And the compilation proceeded with success.
A colleague made the same procedure with success also.

Can you retry in your side with a complete re-installation of the Open STM32 tools? May be a problem occurred during the installation on your computer. And let me know what is the final status.

Thank you
Best Regards

St3dPrinter, 2016-10-11:
Oups, I just see your answer.
The page was not correctly refreshed.

I'm happy to see that you have succeed.

shangaoren, 2018-04-29:
What is the configuration that was used to make it work ?
 i'm getting mad trying to get the project compiled