ManuLST/Marlin4ST #3: New version of the 3D Printer board

St3dPrinter, 2016-10-10:
The objective of the thread is to list all features you want to have in the new 3D printer controler board.
Describe the feature and why you need it.

pirat777knj, 2016-10-26:
Hey.
First of all, I want to note the importance of the availability of USB. No microSD, namely USB. It is good that it is on our board.
When printing it is important to control the flow of the filament. In the case of one or two more or less clear. But how to connect monitors with three extruders?
Also good to see on the board up to 7 stepper motor drivers.
On printers with two engines on an axis overdue. This is impossible when using three extruders (

Neoolog, 2016-11-17:
1/16 step...oups, and 12V on stepper only not 24v...double oups...

EFlynn7778, 2016-11-18:
Neoolog , please explain why?   I see this all the time, and nobody has a good reason why we need more than 1/16 microstepping, or to be able to run on voltages higher than 12.    It simply isnt true.

JustAnother1, 2016-11-18:
Higher voltage allows faster heating as with higher voltage you can send more energy trough wires of the same thickness. If you want to heat your bed fast you would need really thick wires with 12V. With 24V you get double the energy through the same wires.

More then 1/16 stepping helps with stepper noise reduction. 1/16 is currently used on printers. This thread asks for feature for the next generation board. More precise stepping might come in handy for some new designs.

EFlynn7778, 2016-11-18:
@JustAnother1   Sorry, you did not read the question.   I understand the benefit of higher voltage for the heaters.  The reply was about the motors.  :)  I know the answers to my questions, but everybody always argues for higher motor voltage and microstepping with no valid information to back up why.  they just think higher is better , which is not true.

There is not much of a benefit to running the motors of the type of machines this will be used on at higher than 12V , and can actually have a negative impact since the board does not have phase output diodes.

I disagree abut 1/16 microstepping too.  This is a common misconception.   Higher microstepping DOES NOT = greater precision.   Nor does it necessarily translate to quieter operation.   Noisy operation usually comes from motor mismatch to the driver, and resonant coupling to the machine frame.  Microstepping can be a band aid to this problem , but not the solution, and if the machine was designed properly in the first place , it would not be an issue.

@St3dPrinter  Please refrain from taking technical advice from the RepRap community.  Yes, its a great community , and there is a lot of knowledge within it , but almost nobody understands the technical aspects of driving stepper motors properly.

Also, I would like to see output diodes on the motor phases.  This will improve microstepping performance even at 1/16 step.

JustAnother1, 2016-11-18:
@EFlynn7778 I might again be misunderstanding what you are saying, but I don't see the " valid information to back up" your points in your reply.

@St3dPrinter I think what he wanted to say is that you should not blindly implement every feature that someone requests. As that would result in a lot of features that will be of no use.

I agree with EFlynn7778 that there is much knowledge in the RepRap community. If you base your decision on the valid arguments brought forward by those competent members, and ignore the rest, you should be fine.

Regarding stepper motor voltage and micro stepping. I think one can only discuss this with also looking at the stepper controller chips. Then it also depends on what kind of stepper you are intending to use. There are not only PrusaI3's out there. Some people build really big printers and therefore the requirements regarding the stepper parameters might not be the same for everybody.

@EFlynn7778 You might want to take a look at http://www.trinamic.com/products/integrated-circuits/details/tmc260a-pa/
That chip offers up to 40V and 1/256 micro stepping. Can you explain me why they are wrong?

Regarding noise: Part of the reason for too much noise from the stepper is also the firmware. Marlin for example has this feature of doing up to 4 steps in one ISR. This means that the timing of the steps is completely wrong. As some time later the next group of 4 steps will be made.

I agree that it is better to fix the problem then to compensate for the consequences. But sometimes that is not possible, or not the best way to do it. So silent steps are not the killer feature that everybody must have. But if it comes with the stepper chips you use anyway than that is a plus.

St3dPrinter, 2016-12-07:
Thank you for these informations. They are realy usefull.
I will study these points and I will give you, in this thread, a feedback.

EFlynn7778, 2016-12-07:
@JustAnother1  I am not saying there arent applications that require higher voltages and microstepping.  However , the vast majority of machines this controller will be used on, will fit well within the 12v, 1/16th step spec.   

Microstepping is intended to be used to account for various mechanical advantages/disadvantages.  NOT to make your machine run quieter.  NOT to make it more precise.  

Finer microstepping does not necessarily equal more precise movement, and in the case of the RepRap community , higher voltage/microstepping is most often used as a bandaid to poor machine design rather than from genuine necessity.  ST isnt building a Ramps board here , but rather a nice controller that should be paired with nice hardware that is properly designed and doesnt require a bunch of nonsense band-aiding.  If you want a machine equivalent to the Ramps , then fine, continue building poorly designed Prusa's that require poor stepper driving decisions to be made in order for them to operate satisfactorily.

Also remember, steppers are current devices , not voltage , meaning it is current applied to the windings , not voltage , that provides torque.  Do you think that just because you are supplying say 40V to a stepper driver , that is what is being applied to the phase windings?   You would be incorrect.  Ohms law still applies here. Look at some motor spec sheets and figure out based on winding specs, what voltage would be applied to the motor phases to provide full holding torque.

Some basic ( in now way complete/definitive) info that may help you understand better  https://en.wikipedia.org/wiki/Stepper_motor

JustAnother1, 2016-12-07:
@EFlynn7778  I think it is very sad that you don't share your wisdom with us. Just claiming that other statements are false doesn't help here. (I know Owns Law)

I also don't understand your reasoning. You argue that there is a use case for Higher Voltages 

>  I am not saying there arent applications that require higher voltages and microstepping. 

You then argue that the ST board is not intended for only normal use (like ramps on those 12V machines):

>  ST isnt building a Ramps board here , but rather a nice controller 

So I agree with you that this board should not focus only the machines that

>  will fit well within the 12v, 1/16th step spec.

I think you also made clear that you don't need more than 12V and 1/16 micro stepping. Problem is:

> The objective of the thread is to list all features you want to have in the new 3D printer controller board.

12V and 1/16 micro step is already covert in the current version. So just use that or start listing the features that you want.

Neoolog, 2016-12-07:
Just leave the choice to user....if some body is happy with 12v and 16 step...Goood   im happy to. but leave the possibility to other people to do what they want to do with the board and very body will be happy....ohms law or not....

dtt0213, 2017-01-07:
EE here. Please support 24V operation so the steppers can be run at a higher top speed.

EFlynn7778, 2017-01-07:
EE here as well, and I'm not trying to create arguments, but you simply
stating 24v to support higher top speed is not a very accurate statement.
Higher voltage does not necessarily equal higher speed.

dtt0213, 2017-01-07:
Please argue with this:

V/L = di/dt

where:
V = voltage
L = inductance
di/dt = change in current

di/dt represents motor speed. You increase it by either increasing voltage or reducing inductance

EFlynn7778, 2017-01-07:
Not arguing with that at all, but please read some datasheets for typical
motors, and drivers used on these machines. Its not as straight forward as
that.

dtt0213, 2017-01-07:
@St3dPrinter

+1 for adding software to read from a USB memory stick 

Please also consider supporting a TFT with 4.3" touchscreen similar to your STM32F7 disco board. At least add a 40 pin TFT connector.

These additions would make the demo feature set similar to that of the Ultimaker 3 which is based on a much more complicated embedded controller + Linux system. This would make the ST solution a clear differentiator.

JustAnother1, 2017-01-08:
@dtt0213 Do you have details on the Ultiaker3 Schematic (Linux System) ? I only found the Board with the two AVRs,..

dtt0213, 2017-01-08:
I just read about the Ultimaker 3 electronics architecture here:

https://ultimaker.com/en/community/23401-inside-the-ultimaker-3-day-4-electronics

JustAnother1, 2017-01-08:
@dtt0213 thanks for the link.

I like the Ultimaker design. Therefore this suggestion for a new board:

A board without processor that can be connected to a Nucleo would be nice. The Nucleo could then be connected (using UART) to a RaspberryPi/BeagleBone/... Stuff like USB Stick Support, Display, Wifi, Ethernet come for free with the Raspberry.

The board would just be a RAMPS for Nucleo and could therefore be really cheap. With the Nucleos much cheaper than Arduino2560 this would be a great fit for the low end printers. So better performance for less money!

dtt0213, 2017-01-08:
@St3dPrinter

Summary and some more additions:

1) 24V support for faster motor speeds and less current draw for heated beds and heat cartridges. This will reduce wire size and connector specs.

2) Reading USB sticks directly because it's easier to transfer files from a PC versus microSD. Ultimaker3 does this.

3) TFT and touch screen interface for better UI experience. Be sure to include a QSPI flash for storing embedded graphics. SDRAM for vid buffer is also cheap. At minimum provide a connector.

See this vid on a printer that has a nice UI. It also uses an ST chip!
https://www.youtube.com/watch?v=3PstOUPbMP8

4) Camera interface for capturing snap shots of the print. Make pics available on the embedded webpage so users can check up on their prints. At minimum provide a connector.

5) Secondary shutoff circuitry for the extruders and heated beds in case of an MCU crash or firmware bugs. 

I believe all of these features could be added using an STM32F7 chip. Please offer a board/design that is a substantial differentiator from what's out there otherwise it won't gain much traction.

brdirais, 2017-01-14:
Hi,
The EVAL-3DP001 board can be connected to a Raspberry Pi platform. It is what we did. On the RPi, we use Octoprint. The result is really good.
For my point of view, it is important to separate the two world:
- On one side the real-time platform that is in charge of executing the G-Code command
- On the other side the non real time platform, most of time a Linux system, that is in charge of managing the G-Code file, displaying the print progress, taking video, providing wifi access, ...

Neoolog, 2017-01-14:
KEEP IT SIMPLE....stay with a simple board, dont put UI interface on it, just give the possibility to plug a other board ( user choice)  for the UI...Some need display inteface and some not, so the price will increase to much for those who not need inboard UI....KEEP IT SIMPLE

If some chip on the big board broke you lost everithing and need to put big dollard on the table for restart the machine.....KEEP IT SIMPLE

If you modify to much the board it will take long time the produce ......KEEP IT SIMPLE

The firmware will be to big and complex to debug.......KEEP IT SIMPLE

Think MODULAR interface, so the people buy and put what they need for the own  machine so the price still low......KEEP IT SIMPLE

I dont want to pay for another processor  and sdram and display adaptor and camera interface, i got 3 Raspberry and display here for my printer and they do their  job......KEEP IT SIMPLE

GIVE  the USER the CHOICE they need for them, dont force them to have YOUR DREAM BOARD, no body have the same dream.....KEEP IT SIMPLE

brdirais, 2017-01-15:
@Neoolog 
I totally agree with you.
A simple and efficient real time board that just do what we want : execute G-Code
No wifi, no USB master or SD, no display, no keypad.
Just an onboard debugger (ST-Link), a serial link and/or USB link and/or SPI
An extension bus (RPi compatible)
And ... low price

St3dPrinter, 2017-01-23:
Hi,
So, there is 2 main orientations:
- Make a complete standalone board that integrate printing features and also “user interface” features: screen, keypad, SD, USB master, …. . This board will be **in the spirit of the current STEVAL-3DP001 board**:  **Complete & expensive**
[image]

- Make a very simple board that just integrate printing features and an extension board to connect to a µ-computer platform (Raspberry Pi) : **Simple & Cheapest** 
[image]

**Same features as v1 for printing part:**
- 3 extruders (12v and 24v support)
- 1 Heated beds and 1 hot chambers  (12 V \ 24 V)
- Embedded ST-Link V2  (much appreciated by the community)

JustAnother1, 2017-01-23:
I would allow for some "unused" interfaces. So that other stuff can easily be added (Bed Level Sensor, Filament Sensor, RGB LED strip, Flux capacitor,..)

St3dPrinter, 2017-01-24:
It's my mistake. I forget to indicate in the picture the different sensors used for bed leveling (induction, BLTouch or simple switch), temperature sensor,....  
The current board already provides these interfaces. The new board will also do it, independently of the chosen solution.

dtt0213, 2017-01-26:
@St3dPrinter
For the full featured system I would also add USB Host for reading files directly from a USB memory stick. 

When you say the full featured system is "expensive" it may not be compared to a comparable system built on more expensive components like the Ultimaker 3 design. 

From a marketing stand point there are already many low cost designs based on AVR micros flooding the market.  I think there is a void in the higher end boards that the STM32F4 or STM32F7 designs could fill.

BTW - the high end board should be < $100. Think of an STM32F7-DISCO eval board with stepper drivers added. Camera, TFT, Wifi, etc could be accessory items.

JustAnother1, 2017-01-26:
@dtt0213 I agree with you on USB Host. But I disagree on "low cost" vs "expensive". Did you see that the current board is at 135€ ? 
http://de.farnell.com/stmicroelectronics/steval-3dp001v1/ref-design-board-3d-druckertreiber/dp/2627826?ost=3dp&selectedCategoryId=&categoryNameResp=Alle%2BKategorien&searchView=table&iscrfnonsku=false

You are right that there is a lot of competition on low cost solutions. You basically can get everything cheaper from china. And they also start doing their own AVR based boards. But even the simple and cheap solution suggested above will have a ARM Cortex-M CPU. So this is not lowest quality. This does not need to be cheaper than what is currently available.

One of my printer setups is this: A Arduino2560 with a RAMPS 1.4 http://reprap.org/wiki/RAMPS_1.4 connected to a RaspberryPi3 with the Raspi Display (7" touch). On the software side it runs a firmware on AVR and Octoprint on the Pi. With Octoscreen https://github.com/chickenchuck040/octoscreen to have a touch friendly UI.

Ramps 1.4, Pololu(4x), Arduino Mega 25560: 40€
RasPi3 : 35€
RaspiCam: 20€
RaspiDisplay: 80€

Total: 175€

I think that that setup has all the features you want (Ethernet, USB Host, Wifi, TFT, Touch, Camera,..) The high end board you seek should probably have all this but for less money, right?

I can understand Ultimaker in that they don't want to implement drivers on a Cortex-M for all this stuff.

On my example setup replacing the Arduino 2560 with a Nucleo or Discovery board could save some money and also give better performance. The problem is that both boards don't have the Arduino 2560 connectors. All that is needed would be a checkup of the Ramps Schematic to make it 3.3V and a layout that matches the Nucleo pinout. If then the Chinese copy that board and sell it even cheaper that is probably no problem for ST. And with AVR is started the same way, as the obvious next step it so put the ST chip directly on a new board. That is how all those AVR based printer boards were created. But for the community to adopt the Nucleos and therefore the ST ARM chips such a Ramps-ST would be needed. And I think that ST basically wouldn't care if people buy their boards or if all the Chinese boards will use ST chips instead of the (compared to performance) pricey Microchip(previously Atmel) AVR. I think this would be a small thing for ST to do with a potential huge benefit. And nobody says that they can not also build that "expensive sub 100$" board you dream of.

@St3dPrinter One more thing: You might want to think about powering the SBC(Raspi/BBB/or whatever( from the printerbord. (The Raspi can be powered by supplying 5V to the GPIO Pins) This would allow the user to only have on big power supply that powers everything from motors to heaters to the SBC. That also solves common GND problems,...

But that all is just my 2c and my humble opinion. So do whatever you think is best.

dtt0213, 2017-01-27:
@JustAnother1 

I'm taking the OEM perspective- a low cost and integrated design for a full featured capable system. ST does not need to include an LCD, camera, or Wifi on the base kit. They just need to provide headers for these peripherals for future expansion.

I spent the last 2 years designing boards with the STM32F4 and STM32F7 micros with LCD and USB host interface. These are powerful chips. The board I"m talking about can be made for < $100. The STM32F7-DISCO eval board has half of what we need for $50. Spend another $50 for the 3D printer hardware and future expansions.

St3dPrinter, 2017-01-27:
Internally we had also think about a third way that will be a Nucleo expension board, as you suggested.
It will be more a mother board for a nucleo stm32fX board and expension board.
This new board will provide power for the other boards.
It will just (as the simple board) focus on printing

[image]

JustAnother1, 2017-02-02:
@St3dPrinter I just received the "Microcontroller news from STMicroelectronics January 2017" email advertisement. And for the second time they mention the "Arduino STAR OTTO board" When can I buy it and where? The linked Arduino shop seems to not have it,...

The STar Otto would also be a good option as base for the "new expansion board",...

Janusz-M, 2017-03-05:
Hi all.
I got a package from the STEVAL-3D001V1
Joy is large. Nicely done.
I try to run 3D001V1-STEVAL and I have a problem with that.
Very please help in this topic.
I loaded via USB ST-Link binary file from page: https://github.com/St3dPrinter/Marlin4ST
1-on the pcb STEVAL-3D001V1 I set the jumper J22 to ON (enabled)
2-I have connected by ST-Link/USB port ST computer
3-I turned on the power supply 2 x 12V
4-there has been connected STEVAL-3D001V1 with a computer and D30 shone on a permanent basis
5-in the computer opened a new drive F to which I copied a file from https://github.com/St3dPrinter/Marlin4ST/blob/master/Binaries/Marlin.bin
6-there has been a saving to the STEVAL-3D001V1 and after a reboot the file disappeared from drive F
7-I removed the jumper J22 (gaping) she made the USB cable to the mini USB socket in the STEVAL-3D001V1 and turned off power 2 x 12V
8-do not have an SD card inserted in the slot and I do not know whether it is needed?
9-I ran the application: Printrun-Win-Slic3r-03Feb2015
10-preview at: Printrun-Win-Slic3r-03Feb2015 I have:
[ERROR] Printer is not online. Unable to move.
[ERROR] Not connected to the printer.

St3dPrinter, 2017-03-07:
Hi,

The SD card is needed in the firmware version you are trying. It contains the configuration file.
If it is not available, the firmware will stop running, and, of course, no communication with be possible.

But I agree that it is not a good solution to use a SD card.
In my own environment, I de-activate it.

I will make my possible to push this new branch on GitHub for tomorrow evening.

St3dPrinter, 2017-03-08:
@JustAnother1   I will try to get a status regarding the OTTO board.

@Janusz-M   I just pushed the new version of the firmware. You can try it.

Janusz-M, 2017-03-16:
Now I can control the Board with the STEVAL-3D001V1.
I have a message: M999-restart after being stopped with an error
This may be a problem with the data on the SD card.

Whether the LCD display will work?
: http://www.reprapdiscount.com/home/34-full-graphic-smart-lcd-controller.html

He is big and graphic.
Whether in the STEVAL-3D001V1 can control the LCD?

Neoolog, 2017-03-16:
Have already have a M999 with another board. ..it was my temp sensor not working ...

St3dPrinter, 2017-03-30:
@Janusz-M   Do you have technical information of your LCD display?  Is it a I2C display?

Mongooose, 2017-03-30:
@St3dPrinter It looks like all the documentation for the LCD display Janusz-M asked about is available here: [http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller#Schematics](url).

 Is there any decisions or projected time frame for a new design from ST? I am a little late to this party but found the STEVAL-3DP001V1 board and have been debating getting one and working to see what I could do with the firmware.  Although I do already have several discovery and Nucleo boards, so if it will not be long for a new design I might wait.  

 The one main feature I would like to see added is the ability to add atleast a minimal display and rotary encoder for a direct interface to the board, similiar to What Janusz-M linked, better would be an option for a LCD with touch screen.

 Thank you to ST for their interest and looking at what the community wants!

Davey3D, 2017-06-14:
It's well accepted in the industry that higher voltage gives better stepper performance. In terms of heaters, higher voltages = less current = less I2R losses (aka heat) A 24v 5a heater is much easier to deal with than an equivalent power 12v 10a heater. Smaller wires, more common connectors, smaller PCB traces, smaller MOSFETS, less heatsinking. Google how many DIY printers have fried due to overheated wires and connectors.

from oreintalmotor dot com website. They make motors...they should know. :-)
"Driver voltage plays a large part in high speed performance. The higher the ratio of drive voltage to motor voltage, the better the high speed performance. High voltages force the current into the windings at a faster rate than the 63% mentioned above."

Davey3D, 2017-06-14:
Would this work on a regular ST32 dev board like a Nucleo? The STEVAL-3DP001V1 is an impressive board but overkill for my needs.

drepou, 2017-07-20:
I m really looking for a solution to drive the Ramps board with the powerfull f7 disco board but reading the comment here it look like the board is not Ramps compatible rights ? I agree with justanother1 USB/wifi touch screen and fast cpu are mandatory. It will be great if st will make Shields to use the existing board like f7 disco who already have all Logic and graphic to drive à 3D printer just put all the motors drivers and extrudeur on the Shields so if something burn you can change it instead of Trash a 150€ single board

G-Fource, 2018-05-10:
So firstly yes more voltage is better when driving a stepper motor its got to do with the slew rate and how fast your stepping the motors. The faster you step or pulse the motors the higher the voltage needs to be. if the voltage is to low when increasing speed you will lose steps this is because of the voltage rolling off to quickly by the time the pulse has turned off the out put voltage gets clipped and you don't have enough voltage/current to drive the  coils.. When you increase the voltage when you pulse the stepper at a high frequency the voltage will rise quicker and will not get clipped as voltage is increased its rise time is decreased  . Think voltage as pressure the more pressure you have the faster you can actuate a pneumatic cylinder.. these chips you buy to control a stepper are chopper circuits.. the current is measured on the h-bridge so that the power source is turned off if current gets to high.. also if you to gain something there are always losses if you increase the steps to 1/16  or any other increase you will loose torque... and yes the lower the steps per rev the noisier the unit gets and will also vibrate the 3d printer witch would not be good....