ManuLST/Marlin4ST #6: switches donot work

samiaruda, 2017-01-24:
Hi, I have bought an ST board for evaluation.
we have wired in power supply and had it up and running. we used repetier host as our terminal software which connected smoothly and started reading what comes out of the board. we started doing regular checks on components.

Problem1:
first we started with switches; x and y worked but z didnot. it was not about wiring. we went into pin files and had to change input output signals inside the framework, it was reversely defined only on that port we used for z switch. all the rest of the ports was fine, somehow.

we managed to get z switch working by rewriting the strings.

we checked heating elements. they did work.

Problem 2:
we started checking motors. none of them worked. we started measuring potentials through driver connector legs together with processor connections. for some reason drivers didnot get signal when triggered via commands from terminal software. while we were trying to figure out which leg doesnot receive signal by measuring, I made a short circuit and burnt the driver (probably processor also).

Now I have ordered a new board to continue testing.
But, I am afraid of getting it burnt again. Therefore, I am writing to see if someone had faced a similar problem and can share a solution for it. Maybe ST people may provide a solution for this?

Looking forward to your help,
arda

St3dPrinter, 2017-01-27:
I really sorry to see that you faced these issues.

Regarding the Problem n°2, are you sure the temperature sensors was correctly connected? If no, motors don't move.
What is the version of the firmware you used?

samiaruda, 2017-02-12:
hi,
we used the one before the latest, we did the testing on september and downloaded that version at that time.

temperature sensors were working actually, we got readings and they were correct

huixiesui, 2017-12-07:
Hi,
    Try sending this instruction.  //* M401 - Lower Z probe if present
    or// G28 - Home one or more axes